stm32-rs/stm32f7xx-hal #6: Integrating the changes from musitdev's fork

hannobraun, 2019-06-07:
There's a [fork of this repository](https://github.com/musitdev/stm32f7xx-hal/), maintained by @musitdev for his own use, that contains some features not present in this repository. I'm currently working on making some improvements there, and as part of this project, I'm trying to sync up both repositories to get them closer together.

(My work on that repository is the reason for my pull requests this week, by the way. Thank you @mvertescher, for reviewing and merging them so quickly!)

@mvertescher: I've opened this issue to inform you about those features and summarize the state they're in, in case you want to include them in this repository. Since those features are purpose-built to @musitdev's needs, they are not necessarily in a shape that allows them to be submitted here. 

I've rebased `musitdev/master` on top of `mvertescher/master`: https://github.com/mvertescher/stm32f7xx-hal/compare/master...braun-embedded:master

These are the main changes, and what I plan to do about them:
- **Oversampling configuration for USART:** What I'm observing doesn't match the documentation, and I don't understand why. I have a [WIP branch](https://github.com/mvertescher/stm32f7xx-hal/compare/master...braun-embedded:oversampling) (see commit messages for more info). I plan to take another look at this today, but if I can't get it to work quickly, I'll move on to the next task.
- **DAC:** I haven't looked into this one, and don't currently plan to.
- **SPI:** I'll take a closer look at this next week. Depending on how it goes, I'll either submit a pull request, or update this issue with the reason I'm not submitting one.

hannobraun, 2019-06-07:
Update on USART oversampling: Turns out I'm blind and need a debugger to tell me what's obvious. Anyway, problem solved, pull request here: https://github.com/mvertescher/stm32f7xx-hal/pull/7

mvertescher, 2019-06-07:
Thanks for the heads up @hannobraun and thanks for your work on bringing the repositories closer together!

> in case you want to include them in this repository.

Yes! It would be ideal if we could move to a single repository if possible. This crate is currently incomplete and it would be great to integrate @musitdev's work.

> __DAC__: I haven't looked into this one, and don't currently plan to.

I'll create a separate issue for this.

hannobraun, 2019-07-12:
As of #17, the SPI module should include all features from @musitdev's fork (although there is a bug; fixed as part of #18). This only leaves DAC, and there's a separate issue for that. Closing.

closed 2019-07-12 by hannobraun