stm32-rs/stm32f7xx-hal #22: Future plans, updates

no111u3, 2019-11-01:
Your implementation of HAL for stm32f7xx so good but:
- I not found any version on crates.io: which plans to release your hal?
- I have my version with ahead implementation for some featues - https://github.com/no111u3/stm32f7x7-hal, and I ready to integrate my features to your crate.

I'm so very interesting of use your HAL version but I need answer to upper questions.

mvertescher, 2019-11-02:
Thanks for your interest @no111u3 ! :)

> I not found any version on crates.io: which plans to release your hal?

I've created #24 to track the initial release to crates.io. I should be able to take care of that in the next day or so.

> I have my version with ahead implementation for some features - https://github.com/no111u3/stm32f7x7-hal, and I ready to integrate my features to your crate.

:+1: awesome crate. It would be great if we could integrate your work into `stm32f7xx-hal`. I believe I have a stm32f7x7 board so I should be able to help with testing as we add support. Looking forward to some PRs!

no111u3, 2019-11-02:
@mvertescher very thanks for answer.
I'm so have many interests of use community support package. I have many boards on stm32f7xx mcu family - one on stm32f767 and two on stm32f746, and I have plan to buy other in this mcu family. 
So we can create issue for primary features to support this hal.

7h3qu1rkyb1t, 2020-12-01:
Is it still an issue? Else we can close it right? To keep it clean

mvertescher, 2020-12-01:
Yes, this is resolved.

closed 2020-12-01 by mvertescher