stm32-rs/stm32f7xx-hal #34: Timer Makes Incorrect assumption about Clock Speed

dtjones190, 2020-03-04:
The Timer assumes that the clock speed is equal to the peripheral clock, but that is only true in certain circumstances.
See Page 154 https://www.st.com/content/ccc/resource/technical/document/reference_manual/group0/96/8b/0d/ec/16/22/43/71/DM00224583/files/DM00224583.pdf/jcr:content/translations/en.DM00224583.pdf

[image]

rfuest, 2020-06-23:
The issue has already been fixed in #36. I've just tested this again after the changes made in #50 and it seems to work as expected, so I think that this issue can be closed now.

hannobraun, 2020-06-24:
Thanks for confirming this, @rfuest!

closed 2020-06-24 by hannobraun