stm32-rs/stm32f7xx-hal #54: Update to new features make more compatibility with stm32f4xx-hal

no111u3, 2020-06-29:
I found more update version of library - https://github.com/dtjones190/stm32f7xx-hal.
Could you ask me about plans to update with common state such as stm32f4xx-hal, I can create PR based on upper changes.

mvertescher, 2020-06-30:
Cool! I'm happy to review and merge any new functionality or fixes if folks open up PRs. The `stm32f4xx-hal` is certainly more complete than this crate ATM and I agree it would be good to keep things as similar as possible. PRs are definitely welcome to make this crate more compatible with the other stm32 hals!

no111u3, 2020-07-02:
Create first PR with little cosmetic changes #57

no111u3, 2020-07-04:
Add PLL Q configuration for usb #59

no111u3, 2020-07-08:
@mvertescher could You review #59 and #60? - first unblock me for adding other mcu blocks support (usb, sdio, sai and rnd depends on 48 MHz clock source), second provide me more options for test usb stack for stm32f7 mcus.

mvertescher, 2020-07-08:
Np @no111u3, I'll move forward with reviewing/merging asap. I've pinged some folks as well to get any final feedback.

no111u3, 2020-08-31:
@mvertescher I rewrite RCC configuration #74: to work version with highly compatible to previous version, so It add configuration for configure USB clock with or without use PLL for system clock. (It fully replace PR #59). Could you review it. Thanks.

mvertescher, 2020-09-02:
Thanks for the update, I'll take a look!