stm32-rs/stm32f7xx-hal #55: Release 0.2.0

mvertescher, 2020-06-30:
It's been a while since 0.1.0 and it's time to release an updated version. Perhaps use `cargo-release` and add a change log too?

hannobraun, 2020-06-30:
For what it's worth, I used `cargo-release` in the past on various projects and stopped doing so. I don't recall the exact reasons, but I remember I didn't like a lot of things about it. That's probably too vague to be useful, but it's what I got :-)

One thing that I do remember is that it immediately updates the version again after the release. I consider it an anti-pattern to change the version again until right before the next release, as it will prevent the use of `[patch]` to override a dependency on the crate doing that (the `[patch]`ed version needs to be the same as the one depended on, which is annoying enough of a restriction on its own).

mvertescher, 2020-07-01:
Yeah, I've only used `cargo-release` briefly and was a little surprised about some of the operations. However, it did seem convenient. The `[patch]` version mismatch does seems like an issue though. I believe `cargo-release` will up the version automatically to something like `0.2.0-alpha.0` by default, but it looks like this can be disabled with the `--no-dev-version` flag.

Eh, I don't feel strongly about using it though. It's no problem to manually commit/tag/publish/etc this release, so I'll opt to do that.

mvertescher, 2020-07-01:
v0.2.0 is now released! :tada:

closed 2020-07-01 by mvertescher

hannobraun, 2020-07-01:
To be clear, I didn't mean to argue against using it. I just meant to leave my own experience so you can make a more informed decision.

mvertescher, 2020-07-01:
Ah ok, no problem! I did find your thoughts helpful :). Thanks for your input @hannobraun, always welcome!