stm32-rs/stm32f7xx-hal #58: Create generation memory.x for different mcu's in family

no111u3, 2020-07-03:
Than I tried build example for stm32fdisco-scree I got some interested message of compilation:
``` cargo build --example stm32f7disco-screen  --features="stm32f746 rt"
   Compiling stm32f7xx-hal v0.2.0 (/Users/nis/Develop/stm32f7xx-hal)
error: linking with `rust-lld` failed: exit code: 1
  |
  <skipped by unnecessary> 
  = note: rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 130048 bytes
          rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 130049 bytes
          rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 130050 bytes
          rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 130052 bytes
          rust-lld: error: section '.uninit' will not fit in region 'RAM': overflowed by 130052 bytes
          rust-lld: error: section '.uninit' will not fit in region 'RAM': overflowed by 130052 bytes
          rust-lld: error: section '.uninit' will not fit in region 'RAM': overflowed by 130052 bytes

error: aborting due to previous error

error: could not compile `stm32f7xx-hal`.

I think we need to use generation for memory.x

hannobraun, 2020-07-03:
There are multiple ways to do this, but here's some example code that could be copied from:
https://github.com/lpc-rs/lpc8xx-hal/blob/master/build.rs#L42-L63

no111u3, 2020-07-03:
> There are multiple ways to do this, but here's some example code that could be copied from:
> https://github.com/lpc-rs/lpc8xx-hal/blob/59b2a9d996570ad1869e172f8d4fed67958ce4c3/build.rs#L42-L63

Yes, sure. I talked about this solution - so If anybody hasn't concerns to this solution I can create PR for tests.

mvertescher, 2020-07-04:
Yeah, this is a great idea. Adopting the `lpc8xx-hal` implementation sounds good to me.

no111u3, 2020-07-07:
@mvertescher @hannobraun I think this issue can be close by success resolve.

closed 2020-07-07 by mvertescher