stm32-rs/stm32f7xx-hal #73: Pac cannot be brought into scope when compiling with the stm32f769 feature

seanybaggins, 2020-08-24:
# Steps to Reproduce
1. Clone the following project
```bash
git clone git@github.com:seanybaggins/blinky-769.git
```
2. Checkout the commit containing the bug
```bash
git checkout a8a02896871f19aaa9de09256ce517179ec934ac
```
3. Follow the instructions within the `readme.md` for opening the environment within a docker container using VSCode.

4. Use Cargo check to get the compiler error
```bash
cargo check \
    --target thumbv7em-none-eabihf \
    --features stm32f769
```

osphost, 2020-09-14:
Yeah I'm also experiencing this problem.
I had to override Cargo and specify the repository directly:

```
[dependencies.stm32f7xx-hal]
features = ["stm32f746"]
git = "https://github.com/stm32-rs/stm32f7xx-hal"
```

rfuest, 2020-09-17:
In the 0.2.0 version the PAC is exported as `device`: https://github.com/stm32-rs/stm32f7xx-hal/blob/4f42876fc520f1ed2b86cd2c67bd1696cf448372/src/lib.rs#L59
This was changed to `pac` to mach the other STM32 HAL crates, but the change isn't included in a release yet. So you will need to change your code to use `device` instead of `pac` or use the git version.