stm32-rs/stm32f7xx-hal #78: Support display of STM32F723E-DISCO

ctron, 2020-09-15:
I have the "STM32F723E-DISCO" and would like to make use of the display.

It looks like this is attached via the FMC, using an ST7789H2 controller.

I tried to get started with the display example in the `examples` folder, but it looks that doesn't work with the stm32f723 feature enabled.

bbrown1867, 2020-09-24:
The display example is written for the STM32F746G Discovery Board, and it is attached via the LTDC peripheral (not FMC). So even if you were able to get it to compile by modifying the feature, it would not work for the FMC display on your board unfortunate. 

The directory `examples/stm32f7disco-screen` should probably be renamed to something like `examples/stm32f7disco-ltdc-screen` or something to specify that.

ctron, 2020-09-25:
> The directory `examples/stm32f7disco-screen` should probably be renamed to something like `examples/stm32f7disco-ltdc-screen` or something to specify that. I actually didn't realize FMC displays existed, that sounds pretty cool.

Yes, that would make it more obvious :grinning: … I dug a little bit into it, and it looks like a lot of stuff is missing. But I will give it a try.