stm32-rs/stm32f7xx-hal #82: PLL isn't enabled automatically

rfuest, 2020-09-25:
Is the RCC HAL implementation supposed to enable the PLL automatically? I didn't have time to investigate myself, but I think it is: https://github.com/stm32-rs/stm32f7xx-hal/blob/caf2196fcfb10a2947b255f0437a4e9afe99000e/src/rcc.rs#L623

But if I try to configure the RCC using the following code the assertions fails and sysclk is 8 MHz:
```rust
let clocks = rcc
    .cfgr
    .hse(HSEClock::new(8.mhz(), HSEClockMode::Oscillator))
    .sysclk(216.mhz())
    .freeze();

assert_eq!(clocks.sysclk(), 216.mhz().into());
```

If I add a `.use_pll()` the configuration works as expected. @no111u3 could you clarify if this is a bug or the intended behavior?

pftbest, 2020-09-25:
This looks like a bug, the CFGR struct should not have #[derive(Copy)]

`use_pll` takes self by value making a new copy and returns it. The original CFGR is not changed.

no111u3, 2020-09-26:
The original CFGR doesn't change because its never needs to change. I fix configuration for PLL in #83. @rfuest thanks for report of issue.

rfuest, 2020-09-28:
Closed by #83.

closed 2020-09-28 by rfuest