stm32-rs/stm32f7xx-hal #86: usb_fs not enabled for NUCLEO-F767ZI with STM32F767ZI

rHermes, 2020-10-09:
Hey!

I am very new to embedded programming, so please excuse me if this question is not relevant.

I have a NUCLEO-F767ZI, which has a STM32F767ZI chip.

This chips seems to me to support both `usb_fs` and `usb_hs`, but it's not enabled on the lines:

https://github.com/stm32-rs/stm32f7xx-hal/blob/master/src/lib.rs#L83-L93

Is this a mistake, or is something more needed to enable this functionality? Also in the `serial_usb` example, there is as feature for `usb_hs`, but I cannot see this mentioned anywhere else. Is this just for future reference?

Thanks for the awesome crate and any answers :smile:

matoushybl, 2020-11-09:
I second this, I manually added the feature gate for this processor and the USB seems to be working.

It seems to be not there just because the it wasn't tested. I'll make a PR if any maintainer clarifies that there is not another problem why this processor wasn't included.

hannobraun, 2020-11-10:
I'm not aware of any problems, but I haven't been involved in that work and might be missing something. If you tested it successfully, I see no reason not to change it.

closed 2020-11-11 by hannobraun

Disasm, 2020-11-20:
@matoushybl 
> It seems to be not there just because the it wasn't tested. I'll make a PR if any maintainer clarifies that there is not another problem why this processor wasn't included.

Exactly this, it wasn't tested. I tested only with F723 and added all the other chips covered by the same Reference Manual as F723.