stm32-rs/stm32f7xx-hal #87: Support for I2S protocol

ostenning, 2020-10-12:
I'm trying to determine if the SPI implementation can support I2S? If not, any plans in the future to add this? 

Thanks!

hannobraun, 2020-10-13:
I know next to nothing about I2S, but to the best of my knowledge, it is not supported. I'm also not aware of any plans to implement it, but of course any contributions doing so would be very welcome.

ostenning, 2020-10-13:
Thanks for the response @hannobraun. When I find the time, I could begin to look into it.
For further reference, the `stm32f429-hal` [implements it](https://docs.rs/crate/stm32f429-hal/0.0.0/source/src/i2s.rs).

Kind regards, 
Oliver

ostenning, 2020-11-09:
I've begun implementing I2S support, but I need some assistance regarding the way the HAL currently implements the `rcc::CFGR`.

I'm porting the I2S initialization [C++ function `HAL_StatusTypeDef HAL_I2S_Init(I2S_HandleTypeDef *hi2s)`.](https://devel.rtems.org/browser/rtems/c/src/lib/libbsp/arm/stm32f7x/hal/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_i2s.c?rev=c20847a5ad36cab3b4e63bc3021c606820aed87d#L1316)

This function requires access to:
- system clock switch status, to determine if HSI or HSE is being used
- the configured PLLM value
- the `plli2scfgr` register, specifically the `plli2sn` and `plli2sr` values.

These values are then used to determine the `I2S MCK` value, which is crucial for I2S Master operation.

After calling `constrain()` on `pac::RCC`, I lose access to read these registries. Further the `stm32f7xx_hal::rcc::CFGR freeze` function returns a `Clocks` struct that also cannot help me.

Is there an example somewhere of accessing these registries (read-only) in a safe or unsafe way after constraining?
If not, I could attempt to modify the rcc.rs to allow accessing read-only values, or is there any suggestion on how to handle this situation?  

cc. @hannobraun & @no111u3 

Kind regards,
Oliver

hannobraun, 2020-11-10:
You can always conjure up an instance of `pac::RCC` via `pac::RCC::ptr()`. Or you could just add the required information to `Clocks`. I don't have a strong opinion here.

ostenning, 2020-11-10:
Thanks for _pointing_ me in the right direction :+1:

andybest, 2021-04-07:
@ostenning Did you get anywhere with the I2S implementation? I am thinking of possibly implementing it myself if not :)

ostenning, 2021-04-08:
@andybest Sorry, no I ended up going in a different direction with my project. But take a look at the `stm32h7xx-hal`, its already implemented there.

andybest, 2021-04-08:
@ostenning No worries! Thanks, I'll see if I can port over the implementation from there :)