stm32-rs/stm32f7xx-hal #117: A new version on crates.io

dutchmartin, 2021-04-26:
Dear maintainers,

Thank you for creating this awesome crate, it really is a pleasure to work with.
But there is one thing that would working with this crate even more pleasant, and that is releasing a new version on crates.io
Currently some features like the [random](https://github.com/stm32-rs/stm32f7xx-hal/pull/103) implementation are not released yet due to this, and that is unfortunate, since it means that your awesome new work is only usable by setting the git link or downloading it.

So can you please release a new version on crates.io?

If I can help with this in any means (testing the examples on my 767ZI board, or submit a PR), please let me know.
Martijn

hannobraun, 2021-04-26:
Hey @dutchmartin, thanks for opening this issue. Looks like I don't have the power to make new releases. @mvertescher, can you take care of this?

adamgreig, 2021-04-26:
By the way, a few of the other HALs have added their stm32-rs org team to the crates.io owners, so that anyone on the team can publish to crates.io -- there's also an stm32f7 team (with just @hannobraun and @mvertescher), so potentially that could be added to make this easier in the future: `cargo owner --add github:stm32-rs:stm32f7`.

mvertescher, 2021-04-26:
Awesome, that sounds great @adamgreig. I've updated the owners with the `github:stm32-rs:stm32f7` team so @hannobraun can deploy in the future.

Yeah, it's been a while since the last release. I've just pushed `v0.3.0` to crates.io. 

Feel free to open any additional issues for bugs you may find @dutchmartin, that'd be very helpful :)

closed 2021-04-26 by mvertescher

dutchmartin, 2021-04-26:
Thank you for your very quick response!